gbmhunter/NinjaTerm #210: "Clear Text" action causes TX to be buffered, but not shown when "Separate TX/RX" option is selected

merberich-axonvr, 2017-09-23:
On Windows 10, latest update

Conditions:
1) Open a fresh instance of Ninjaterm.
2) Open any COM port at any baud.
3) On the RX/TX pane, select "Display -> Layout Options -> Separate TX/RX" if it is not already selected. Bringing context to the TX area and typing should now show the buffered text.
3) Press "Clear text". Any TX text that was buffered should be cleared.
4) Attempting to enter new text in the TX area should no longer appear. The characters are still buffered appropriately (confirmed with attached Serial devices on my side), but are not displayed in the TX area.

My guess is that this is just some flag that exists on a per-tab basis which is permanently set when "Clear Text" is pressed.

gbmhunter, 2017-10-07:
Awesome, thanks for the bug report! I will look into fixing this other the course of the next few days (it's a long weekend!).

gbmhunter, 2017-10-09:
Fixed! It was a little more complex than a flag, the text is rendered inside a WebView (HTML + JS) for rich text functionality. The "Clear Text" button was incorrectly removing all child <span> elements from the text area, which happened to also delete the caret (if enabled). Once the caret `<span>` was destroyed, adding new text did not work correctly as it was trying to add the text immediately before the non-existent caret `<span>`.

closed 2017-10-09 by gbmhunter

merberich-axonvr, 2017-10-09:
Excellent, thanks!

gbmhunter, 2017-10-09:
If you have run the latest update (v0.8.10), did it get stuck on the splash screen? It does on my computer, I'm currently trying to work out why these changes would cause this....

merberich-axonvr, 2017-10-09:
Yep, I'm stuck on the splash screen as well.

commit 4c95ad7f (2017-10-09): Fixed bug where clicking the 'Clear Text' button then stopped TX text from being displayed, closes #210.