gbmhunter/NinjaTerm #212: Fix bug where NinjaTerm freezes on splash screen when running .exe in Windows.

gbmhunter, 2017-10-09:


merberich-axonvr, 2017-10-09:
I suspect this might have to do with project configuration; if I run the JAR file instead of the Windows executable, the program runs (with the update for cleared text) but is unable to identify the version.

I did notice a change to this line:
`<component name="ProjectRootManager" version="2" languageLevel="JDK_1_3" default="false" project-jdk-name="1.8" project-jdk-type="JavaSDK">`

Perhaps that caused an issue with the updater (JDK incompatibility) and thus the executable only?

gbmhunter, 2017-10-09:
Ah yes it could be that! That line was changed by IntelliJ IDEA when I updated from a 2016 to 2017 version, I might have to specify JDK_1_8 or higher.

gbmhunter, 2017-10-09:
Aha, it's due to the version number having one extra character in it now when it went from 0.8.9 to 0.8.10:

`2017-10-09 15:54:24,252 DEBUG [JavaFX Application Thread] n.m.n.Main [Main.java:148] java.specification.version (as double) = 1.8
2017-10-09 15:54:24,337 DEBUG [JavaFX Application Thread] n.m.n.v.s.SplashScreenViewController [SplashScreenViewController.java:174] SplashScreenViewController() called.
2017-10-09 15:54:30,854 ERROR [JavaFX Application Thread] n.m.n.Main [Main.java:286] java.lang.ArrayIndexOutOfBoundsException: 53
	at ninja.mbedded.ninjaterm.view.splashScreen.SplashScreenViewController.createNextKeyFrame(SplashScreenViewController.java:250)
	at ninja.mbedded.ninjaterm.view.splashScreen.SplashScreenViewController.lambda$createNextKeyFrame$3(SplashScreenViewController.java:271)
	at com.sun.scenario.animation.shared.TimelineClipCore.visitKeyFrame(TimelineClipCore.java:239)
	at com.sun.scenario.animation.shared.TimelineClipCore.playTo(TimelineClipCore.java:180)
	at javafx.animation.Timeline.impl_playTo(Timeline.java:176)
	at javafx.animation.AnimationAccessorImpl.playTo(AnimationAccessorImpl.java:39)
	at com.sun.scenario.animation.shared.SingleLoopClipEnvelope.timePulse(SingleLoopClipEnvelope.java:99)
	at javafx.animation.Animation.impl_timePulse(Animation.java:1102)
	at javafx.animation.Animation$1.lambda$timePulse$25(Animation.java:186)
	at java.security.AccessController.doPrivileged(Native Method)
	at javafx.animation.Animation$1.timePulse(Animation.java:185)
	at com.sun.scenario.animation.AbstractMasterTimer.timePulseImpl(AbstractMasterTimer.java:344)
	at com.sun.scenario.animation.AbstractMasterTimer$MainLoop.run(AbstractMasterTimer.java:267)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:506)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:490)
	at com.sun.javafx.tk.quantum.QuantumToolkit.lambda$runToolkit$404(QuantumToolkit.java:319)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Unknown Source)`

gbmhunter, 2017-10-10:
Fixed in 617887a48749a7106346ace9f1c9b62a4d67a4cb which is part of the new release [v0.8.11](https://github.com/mbedded-ninja/NinjaTerm/releases/tag/v0.8.11).

closed 2017-10-10 by gbmhunter