gbmhunter/NinjaTerm #215: Exception: ReferenceError: Can't find variable: numCharsToRemove

merberich-axonvr, 2017-10-10:
Hmm, found another one. I think it's related to the number of characters tracked in the RX pane. When I hit the exception I was reading a large amount of input characters, and I'm able to reproduce the error under similar conditions. I haven't hit a RX pane maximum before this release, so I think a recent change caused this. (Sorry for all the trouble)

```
netscape.javascript.JSException: ReferenceError: Can't find variable: numCharsToRemove
	at com.sun.webkit.dom.JSObject.fwkMakeException(JSObject.java:128)
	at com.sun.webkit.WebPage.twkExecuteScript(Native Method)
	at com.sun.webkit.WebPage.executeScript(WebPage.java:1455)
	at javafx.scene.web.WebEngine.executeScript(WebEngine.java:982)
	at ninja.mbedded.ninjaterm.util.javafx.comDataPaneWeb.ComDataPaneWeb.runScriptWhenReady(ComDataPaneWeb.java:460)
	at ninja.mbedded.ninjaterm.util.javafx.comDataPaneWeb.ComDataPaneWeb.trimIfRequired(ComDataPaneWeb.java:424)
	at ninja.mbedded.ninjaterm.util.javafx.comDataPaneWeb.ComDataPaneWeb.addData(ComDataPaneWeb.java:340)
	at ninja.mbedded.ninjaterm.view.mainWindow.terminal.txRx.TxRxViewController.lambda$Init$12(TxRxViewController.java:363)
	at ninja.mbedded.ninjaterm.util.rxProcessing.rxDataEngine.RxDataEngine.parse(RxDataEngine.java:396)
	at ninja.mbedded.ninjaterm.model.terminal.txRx.TxRx.addRxData(TxRx.java:303)
	at ninja.mbedded.ninjaterm.model.terminal.Terminal.lambda$handleOnRxData$1(Terminal.java:150)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
	at java.lang.Thread.run(Unknown Source)
```

gbmhunter, 2017-10-10:
Ohhh I have a hunch that this is because I added:

````
"use strict";
````

to the top the the Javascript file, which forces all variables to be declared before you can use them, but I missed putting `var` in front of `numCharsToRemove` (this error will only occur at runtime, when the `trim()` function is called, when there is quite a bit of data). Will try and fix in the next day or so.

This is no trouble! Thanks for helping improve the software!

gbmhunter, 2017-10-18:
Bug has been fixed in https://github.com/mbedded-ninja/NinjaTerm/releases/tag/v0.8.12

commit 7e433ceb (2017-10-17): Fixed bug where exception 'ReferenceError: Can't find variable: numCharsToRemove' was being thrown when a large number of chars were sent to NinjaTerm, closes #215.
closed 2017-10-18 by gbmhunter