gbmhunter/NinjaTerm #218: New logo for NinjaTerm

reallinfo, 2018-03-26:
I wanted to contribute to Ninjaterm. I designed a logo for Ninjaterm. If you like it, I'll send you the files.

[image]

gbmhunter, 2018-03-26:
Wow dude that looks awesome! Yes I would be very happy to use that as the logo for NinjaTerm. If you send me the files, I can update the logo in the website and in the app, as well as give you a mention for that. Do you have a website/name you wish me to add as a thanks?

reallinfo, 2018-03-26:
I'm glad you like it. 

Here are all the files;

https://drive.google.com/open?id=1YH1VhoNWgEf_voiN9NAKgRQ2w36k6dc8

I do not have a website but you can write ''thanks utopian''  it here :)

gbmhunter, 2018-03-26:
O.k. will do! It might take me a number of weeks (maybe a month or two) before I get around to doing this (busy at the moment), but I won't forget!

gbmhunter, 2018-05-18:
Done! (as of v0.9.0).

closed 2018-05-18 by gbmhunter