gbmhunter/NinjaTerm #225: Crash on backspace in TX view

ijager, 2018-12-28:
When pressing backspace if the TX view is in focus the app crashes:

```
java.lang.StringIndexOutOfBoundsException: String index out of range: 0
	at java.lang.String.charAt(String.java:658)
	at ninja.mbedded.ninjaterm.model.terminal.txRx.TxRx.handleKeyPressed(TxRx.java:157)
	at ninja.mbedded.ninjaterm.view.mainWindow.terminal.txRx.TxRxViewController.handleKeyTyped(TxRxViewController.java:652)
	at ninja.mbedded.ninjaterm.view.mainWindow.terminal.TerminalViewController.handleKeyTyped(TerminalViewController.java:253)
	at ninja.mbedded.ninjaterm.view.mainWindow.terminal.TerminalViewController.lambda$init$4(TerminalViewController.java:161)
	at com.sun.javafx.event.CompositeEventHandler$NormalEventFilterRecord.handleCapturingEvent(CompositeEventHandler.java:282)
	at com.sun.javafx.event.CompositeEventHandler.dispatchCapturingEvent(CompositeEventHandler.java:98)
	at com.sun.javafx.event.EventHandlerManager.dispatchCapturingEvent(EventHandlerManager.java:223)
	at com.sun.javafx.event.EventHandlerManager.dispatchCapturingEvent(EventHandlerManager.java:180)
	at com.sun.javafx.event.CompositeEventDispatcher.dispatchCapturingEvent(CompositeEventDispatcher.java:43)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:52)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
	at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
	at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
	at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
	at javafx.event.Event.fireEvent(Event.java:198)
	at javafx.scene.Scene$KeyHandler.process(Scene.java:3964)
	at javafx.scene.Scene$KeyHandler.access$1800(Scene.java:3910)
	at javafx.scene.Scene.impl_processKeyEvent(Scene.java:2040)
	at javafx.scene.Scene$ScenePeerListener.keyEvent(Scene.java:2501)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:217)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler$KeyEventNotification.run(GlassViewEventHandler.java:149)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleKeyEvent$352(GlassViewEventHandler.java:248)
	at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:389)
	at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleKeyEvent(GlassViewEventHandler.java:247)
	at com.sun.glass.ui.View.handleKeyEvent(View.java:546)
	at com.sun.glass.ui.View.notifyKey(View.java:966)
```

Same error for both empty and non-empty strings.

Version 1.1.0
MacOS 10.14.2

gbmhunter, 2018-12-29:
@ijager thanks for the bug report!

Unfortunately, I was unable to recreate the bug. I am running v1.1.0 on Windows. I selected the TX view and tried both backspace and delete, neither caused the app to crash.

[image]

I may be able to get access to a Mac soon to try that...

gbmhunter, 2021-01-02:
Closing as app is now built on Electron, no longer relevant.

closed 2021-01-02 by gbmhunter