gbmhunter/NinjaTerm #226: Home page in Help->About menu is returning a 404 Page not found error.

martinbra, 2020-10-19:
[image]

The above about menu links to http://mbedded-ninja.github.io/NinjaTerm/ which is returning a 404 Page not found error. 
I think that http://gbmhunter.github.io/NinjaTerm/ is the correct one.
Maybe one's could also update the copyright dates.

gbmhunter, 2020-10-27:
@martinbra thanks for the observation! In all honesty, I haven't given this project the love it needs in the last few years! I see you've forked the repo and made the fix, if you submit a pull-request I'll approve it to get in merged into this repo if you want.

martinbra, 2020-10-27:
I tried a pull request but it didn't pass the compilation because I did not compiled locally, updated version, etc. Can you do it?

commit 33f8dc18 (2020-11-01): update website address (fix #226 )
commit f97739ec (2020-11-01): Merge pull request #227 from martinbra/master

Update website address (fix #226 ).
closed 2020-11-01 by gbmhunter