gbmhunter/NinjaTerm #314: Is it possible to send a break statement?

tunguskar, 2024-05-07:
## Summary

I need the option to send a break statement. Is there an option?

gbmhunter, 2024-05-07:
Hi @tunguskar , no NinjaTerm can't do that at the moment, But I think it should!

Having a quick look, luckily I think this is possible via this API: https://developer.mozilla.org/en-US/docs/Web/API/SerialPort/getSignals

How do you think the user should be able to trigger a break? Is there a standard keyboard shortcut that does this? Or a UI button?

tunguskar, 2024-05-07:
> Hi @tunguskar , no NinjaTerm can't do that at the moment, But I think it should!
>
> Having a quick look, luckily I think this is possible via this API: [...]

Hi gbmhunter,

When you prepare a string to send in docklight it is done like this:

`Docklight Scripting supports sending a "break" within a Send Sequence and detecting a "break" state using a Receive Sequence definition. "break" signals are added to your sequence definition by inserting a Function Character '%' (F10 key). A Docklight "break" signal has a minimum length of 15 * <nominal bit length>.`

When you press F10 you get this:

[image]

Hterm does not offer this option.

TerraTerm has this: 

[image]

But i dont like this.

gbmhunter, 2024-05-10:
Thanks @tunguskar , I'll look to add this in the next week or so.

gbmhunter, 2024-05-10:
Hi @tunguskar , I've added in the ability to send a break for 200ms by pressing Ctrl-Shift-B when the port is open and the terminal in focus (you should get a message saying "break sent".

It's currently in a deploy preview at https://deploy-preview-315--ninjaterm.netlify.app/

Are you able to test this for me please?

tunguskar, 2024-05-16:
> Hi @tunguskar , I've added in the ability to send a break for 200ms by pressing Ctrl-Shift-B when the port is open and the terminal in focus (you should get a message saying "break sent".
>
> It's [...]

Hi gbmhunter, you are fast. I will give it a try thanks for investigating

gbmhunter, 2024-05-16:
@tunguskar no worries! I actually also released that to production on the weekend, so https://ninjaterm.mbedded.ninja/ should have the change also :-)

gbmhunter, 2024-06-03:
Added in https://github.com/gbmhunter/NinjaTerm/releases/tag/v4.14.0.

tunguskar, 2024-06-12:
sorry for delay. Did now a test. Works so far. Some comments: 

[image]

When sending break I get this messages:

[image]

I get this orange message due to the fact that the target also sends back a break statement for a line ending. So it is not an error it ist ok from my side. 

Maybe you can add this break statement via checkbox, that it is automatically added after payload.

Kind regards

gbmhunter, 2024-06-13:
@tunguskar thank for the feedback.

1. You can change to hex view by clicking the settings icon on the left and then going to "RX Settings". Perhaps I should make it so  the app automatically navigates to there is you click the "ASCII" text in the bottom toolbar.
2. Ok, what I can do is add an checkbox which if selected, disables these snackbar warnings when breaks are received.
3. I could make it so "break" is another "on enter" option. Then allow that settings to be used for hex too. That way, a break will be sent at the end of every line in the macro data text field.

Thoughts on this?

tunguskar, 2024-06-13:
> @tunguskar thank for the feedback.
>
> 1. You can change to hex view by clicking the settings icon on the left and then going to "RX Settings". Perhaps I should make it so  the app automatically [...]

1. Ok found it. Your proposal would be great.
2. Sounds also good
3. Good proposal

Thanks for all your effort

tunguskar, 2024-06-13:
> 1. Ok found it. Your proposal would be great.
> 2. Sounds also good
> 3. Good proposal
>
> Thanks for all your effort

in Docklight there is a character you can add to your payload. than the system automatically sends the break. If you want to automate stuff maybe this is the better way. But it would be just another option. 

[image]

gbmhunter, 2024-06-13:
Ahh yup. This would cleanly work for hex as there are many characters which you could choose which are not valid hex. This wouldn't work so well in ASCII mode since you'd need to introduce an escape character or something similar. I'll have a think about this.

closed 2024-06-15 by gbmhunter

gbmhunter, 2024-06-15:
@tunguskar I have added the features and released a new production build. See if it works well for your use case! I ended up adding a separate option just for hex macros to send a break at the end of every line in the macro text box. The checkbox to hide "break received" warnings might be a little tricky to find, it's at the bottom of the RX Settings view.