gbmhunter/NinjaTerm #323: Often crash when tabbing back to terminal from parameters, with a running open line

FredericHerque, 2024-06-08:
(please note that English is not my native language)

Hello,

I'm using ninjaTerm since some days only, it's really far better than terminal or putty when coding !

My case : 
A serial port is opened and running in terminal (it's some gps stream : arduino uno with gps hat and more).

I tab to setting, or logging. Then go back to terminal.
I expect to see terminal output again.

## Current Behavior

App crash with following message :

[image]

## Steps to Reproduce

1. Discovered when searching for serial parameters of the device line
   (show terminal, go to settings, close/change/open port, return to terminal, BAM)

2. Seeing it when tabbing

3. For this issue, i wanted to capture screen. Just going to setting & back. BAM again

## Possible Solution (Not obligatory)

I can't understand such an easily exploding bug can exist. 
So, it seems logical it could be a local problem with my computer.

Ready to offer help & data to investigate even more.

## Your Environment

- NinjaTerm version (displayed in upper right-hand side of window): 4.17.0
- Chrome version : Version 125.0.6422.142 (64 bits)
- Windows 10 last update & so on
- Good old PC (i7 5960X)

gbmhunter, 2024-06-11:
Hi @FredericHerque . Thanks for this detailed bug report! I've seen the same thing happen, very occasionally. I have yet been able to reproduce it well enough to debug the issue. I'll put some more effort into finding out how to reproduce it and/or what the problem could be.

gbmhunter, 2024-06-11:
Hi @FredericHerque , I've created a "test" release of the app with some debug statements which might help us find the cause. Could you see if you can get the problem to occur with the release at https://deploy-preview-325--ninjaterm.netlify.app/ ?

If it happens, let me know what gets printed to the debug console.

FredericHerque, 2024-06-15:
Hi Geoffrey,

Thanks for this test release.

Interestingly, i cannot easily reproduce the issue. 
I wanted to take some traces for you, trying to reproduce issue with a simple good'ol commercial usb GPS receiver. But ninjaTerm is working flawlessly with this device.

So, I now strongly believe that issue only occurs when using this custom gps receiver i previously used.
But, this prototype will not be available again until 24, jun.
At this date, i will retry investigations.

gbmhunter, 2024-06-15:
@FredericHerque were you using the test release or production release when you couldn't reproduce it?

FredericHerque, 2024-06-15:
Very fine instinct, @gbmhunter !

No error with test release, but I can reproduce every time with production release. 
Problem is definetely in transition from setting tab to terminal tab.
Detailled procedure : 
- GPS is 9600 baud
- if settings to 9800,8,N,1 all is ok, any click sequence from tabs, buttons, all is ok.
- going to settings, close port
- set 56700
- click terminal tab => crash
- crash with 19200 also, but not 9601 nor 12000 bauds, 14400 crash etc...
Doesn't like some events associated with bad baud rates ?

I also note framing error sometime displayed when i play with bits number : if i set at 7, ok error. Then i go back to 8, still framing error ... something in memory, not updated ?

gbmhunter, 2024-06-15:
@FredericHerque ok weird, this crash might of been because of some buttons I was disabling inside of tooltips on the profiles view, without adding a "div" wrapper around them. I have fixed this in the latest production release, which I released a few minutes ago (v4.18.0). Hopefully this fixes the problem! Let me know :-)

FredericHerque, 2024-06-15:
Correction seems effective in 4.18.0 !
I try many times to reproduce issue, but no crash anymore.
Well done :)

gbmhunter, 2024-06-16:
No problem, thanks for the assistance!

Closed in https://github.com/gbmhunter/NinjaTerm/releases/tag/v4.18.0

closed 2024-06-16 by gbmhunter